tier4/caret #31: rclcpp build fails in docker

kminoda, 2022-09-12:
I encountered a following build error in rclcpp (the similar error around TRACEPOINT continues all along for thousands of lines, and below is just a summary)

```
Starting >>> rclcpp                                                                                                                                                                           
Starting >>> rcl_lifecycle                                                                                                                                                                    
Starting >>> rcl_action                                                                                                                                                                       
Finished <<< rcl_lifecycle [14.8s]                                                                                                                                                            
Finished <<< rcl_action [16.7s]                                                                                                                                                               
--- stderr: rclcpp                                                                                                                                                                            
In file included from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_service_callback.hpp:26,                                                                                  
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/service.hpp:28,                                                                                               
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/callback_group.hpp:25,                                                                                        
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_executable.hpp:20,                                                                                        
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategy.hpp:25,                                                                                       
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategies.hpp:18,                                                                                     
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executor_options.hpp:20,                                                                                      
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executor.hpp:36,                                                                                                               
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:26,                                                                     
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors.hpp:21,                                                                                             
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/rclcpp.hpp:156,                                                                                                                
                 from /root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/test/benchmark/benchmark_parameter_client.cpp:21:                                                                            
/root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_subscription_callback.hpp: In member function _void rclcpp::AnySubscriptionCallback<MessageT, AllocatorT>::dispatch(std::shared_
ptr<_Tp>, const rclcpp::MessageInfo&)_:                                                                                                                                                       
/root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_subscription_callback.hpp:236:5: error: _ros_trace_dispatch_subscription_callback_ was not declared in this scope; did you mean 
_ros_trace_rclcpp_subscription_callback_added_?                                                                                                                                               
  236 |     TRACEPOINT(                                                                                                                                                                       
      |     ^~~~~~~~~~                                                                                                                                                                        
/root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_subscription_callback.hpp: In member function _void rclcpp::AnySubscriptionCallback<MessageT, AllocatorT>::dispatch_intra_proces
s(std::shared_ptr<const _Tp>, const rclcpp::MessageInfo&)_:                                                                                                                                   
/root/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/any_subscription_callback.hpp:288:5: error: _ros_trace_dispatch_intra_process_subscription_callback_ was not declared in this scope 
  288 |     TRACEPOINT(  
```

The environment I am using is as follows:
- Docker
- ROS: galactic
- Ubuntu: 20.04
- Linux kernel: 5.15.0-47-generic
- Python: 3.8.10

Has anyone encountered a similar issue?
It seems that the CARET is only tested in 5.13.x kernels, so it is possible that the issue comes from the kernel version incompatibility?

kminoda, 2022-09-12:
It seems that the failure in `./setup_caret.sh` command caused the above issue. After re-running the script, the build worked fine.

closed 2022-09-12 by kminoda

takeshi-iwanari, 2022-09-12:
@hsgwa 
Does CARET work in Docker?
`ros2 run tracetools status` returns `Tracing disabled` even after restarting Docker.

I encountered the same error in my PC before, but it became "enabled" after restarting in my PC.
Is there any trick to make it "enabled" in virtual environment?