tier4/caret #37: cannot setup CARET on Ubuntu 20.04

Torreskai0722, 2022-09-30:
I have set up all the dependencies for caret but got the following error when building:

```
hydrapc@hydrapc:~/projects/ros2_caret_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
[0.442s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'rclcpp_components' is in: /opt/ros/galactic
	'rcl' is in: /opt/ros/galactic
	'rcl_action' is in: /opt/ros/galactic
	'rclcpp_lifecycle' is in: /opt/ros/galactic
	'rcl_yaml_param_parser' is in: /opt/ros/galactic
	'rcl_lifecycle' is in: /opt/ros/galactic
	'rclcpp_action' is in: /opt/ros/galactic
	'rclcpp' is in: /opt/ros/galactic
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding rcl rcl_action rcl_lifecycle rcl_yaml_param_parser rclcpp rclcpp_action rclcpp_components rclcpp_lifecycle

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> rcl_yaml_param_parser
Starting >>> tracetools
Starting >>> tracetools_trace
Starting >>> tracetools_read
Starting >>> caret_analyze
Starting >>> caret_lint_common
Starting >>> indicators_vendor
Starting >>> Mimick
Finished <<< tracetools [0.75s]                                              
Finished <<< caret_lint_common [0.73s]
Finished <<< rcl_yaml_param_parser [0.82s]                                   
Finished <<< indicators_vendor [0.76s]
Starting >>> rcl
Starting >>> caret_analyze_cpp_impl
Finished <<< Mimick [0.79s]
Finished <<< caret_analyze_cpp_impl [0.18s]                     
--- stderr: tracetools_trace                                    
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.1.36ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.23ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< tracetools_trace [1.34s]
--- stderr: tracetools_read
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.1.36ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.23ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< tracetools_read [1.34s]
Starting >>> tracetools_launch
Starting >>> ros2trace
--- stderr: caret_analyze                                    
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.1.36ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/home/hydrapc/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.23ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
[... 90 lines cut ...]
                 from /home/hydrapc/projects/ros2_caret_ws/src/ros2/rcl/rcl/src/rcl/network_flow_endpoints.c:21:
/home/hydrapc/projects/ros2_caret_ws/src/ros2/rcl/rcl/include/rcl/event_callback.h:18:10: fatal error: rmw/event_callback_type.h: No such file or directory
   18 | #include "rmw/event_callback_type.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rcl.dir/build.make:284: CMakeFiles/rcl.dir/src/rcl/network_flow_endpoints.c.o] Error 1
/home/hydrapc/projects/ros2_caret_ws/src/ros2/rcl/rcl/src/rcl/publisher.c: In function ‘rcl_publisher_wait_for_all_acked’:
/home/hydrapc/projects/ros2_caret_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:328:19: warning: implicit declaration of function ‘rmw_publisher_wait_for_all_acked’; did you mean ‘rcl_publisher_wait_for_all_acked’? [-Wimplicit-function-declaration]
  328 |   rmw_ret_t ret = rmw_publisher_wait_for_all_acked(publisher->impl->rmw_handle, rmw_timeout);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                   rcl_publisher_wait_for_all_acked
make[1]: *** [CMakeFiles/Makefile2:431: CMakeFiles/rcl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rcl [5.42s, exited with code 2]

Summary: 13 packages finished [6.57s]
  1 package failed: rcl
  8 packages had stderr output: caret_analyze rcl ros2caret ros2trace tracetools_launch tracetools_read tracetools_test tracetools_trace
  9 packages not processed

```

It seems the ros galactic rmw has deleted the "event_callback_type.h".

takam5f2, 2022-10-02:
@Torreskai0722  
Thank you for reporting.  
Could you tell me which branch of CARET your are trying?

Torreskai0722, 2022-10-02:
> @Torreskai0722 Thank you for reporting. Could you tell me which branch of CARET your are trying? Are you working on [`galactic`](https://github.com/tier4/caret/tree/galactic) ?
> 
> We'll be glad if you share the procedure to find the same issue. Best,

Yes, I checkout to galactic branch since I'm using Ubuntu 20.04 with ROS galactic.

I followed the setup [here](https://tier4.github.io/CARET_doc/refs-tags-v0.2.3/tutorials/installation/). I manually installed lttng 2.12.5 with this [deb package](https://launchpad.net/ubuntu/+source/lttng-modules) since the 2.12 version cannot be installed.

takam5f2, 2022-10-03:
@Torreskai0722  
I doubt that you have checked out to galactic branch before execution of `vcs import src < caret.repos`.  
`event_callback.h` which appeared in your error message has been added since ROS 2 Humble release ([link](https://github.com/ros2/rcl/tree/humble/rcl/include/rcl)). `event_callback.h` does not exist in ROS 2 Galactic ([link](https://github.com/ros2/rcl/tree/galactic/rcl/include/rcl)).

Could you try the following commands in order?

```
git clone https://github.com/tier4/caret.git -b galactic ros2_caret_ws
mkdir src
vcs import src < caret.repos 
./setup_caret.sh # this is necessary for executing colcon build
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Torreskai0722, 2022-10-03:
> @Torreskai0722 I doubt that you have checked out to galactic branch before execution of `vcs import src < caret.repos`. `event_callback.h` which appeared in your error message has been added since [...]

It works! Thanks for your reply.

Torreskai0722, 2022-10-03:
> It works! Thanks for your reply.

I finished the setup and tried the measurement tutorial.

I can get the results under evaluate/e2e_sample, but got the following error when run check_ctf. I have found both check_ctf.py and record_cpp_impl.py files and changed their permissions. But it doesn't solve the issue. Any idea of how to address it?

Thanks!
```
(base) hydrapc@hydrapc:~/projects/ros2_ws/evaluate$ ros2 caret check_ctf -d ~/projects/ros2_ws/evaluate/e2e_sample/
Failed to load entry point 'check_ctf': No module named 'record_cpp_impl'
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.4', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 39, in main
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 250, in add_subparsers_on_demand
    extension.add_arguments(
  File "/home/hydrapc/projects/ros2_caret_ws/build/ros2caret/ros2caret/command/caret.py", line 22, in add_arguments
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'check_ctf'

```

takam5f2, 2022-10-04:
@Torreskai0722  
I suspect that path of command did not included in PATH variable.

Could you try as below?

```
source /path/to/ros2_caret_ws/install/setup.bash
ros2 caret check_ctf -d ~/projects/ros2_ws/evaluate/e2e_sample/
```

Torreskai0722, 2022-10-04:
> 

I have sourced it. Still doesn't work.
```
(base) hydrapc@hydrapc:~$ source ~/projects/ros2_caret_ws/install/setup.bash
(base) hydrapc@hydrapc:~$ ros2 caret check_ctf -d ~/projects/ros2_ws/evaluate/e2e_sample/
Failed to load entry point 'check_ctf': No module named 'record_cpp_impl'
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.4', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 39, in main
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 250, in add_subparsers_on_demand
    extension.add_arguments(
  File "/home/hydrapc/projects/ros2_caret_ws/build/ros2caret/ros2caret/command/caret.py", line 22, in add_arguments
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'check_ctf'

```

hsgwa, 2022-10-05:
Depends on permittions, similar issue can reproduce.

```
chmod 000 -R ./ros2_caret_ws/install/caret_analyze_cpp_impl/lib
```

```
$ ros2 caret check_ctf -d e2e_sample
Failed to load entry point 'check_ctf': No module named 'record_cpp_impl'
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.4', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 39, in main
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 250, in add_subparsers_on_demand
    extension.add_arguments(
  File "/home/hasegawa/ros2_caret_ws/build/ros2caret/ros2caret/command/caret.py", line 22, in add_arguments
    add_subparsers_on_demand(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'check_ctf'
```

It seems that libraries in `caret_analyze_cpp_impl` may don't have enough permissions.

hsgwa, 2022-10-06:
If caret_analyze_cpp_impl is built correctly, following binaries should exist.

```
$ find install/caret_analyze_cpp_impl/ | grep "\.so"
install/caret_analyze_cpp_impl/lib/python3.8/site-packages/record_cpp_impl.cpython-38-x86_64-linux-gnu.so
install/caret_analyze_cpp_impl/lib/libcaret_analyze_cpp_impl.so
```

Were there any warning or errors in `colcon build`?

```
$ colcon build --symlink-install --packages-select caret_analyze_cpp_impl
Starting >>> caret_analyze_cpp_impl
Finished <<< caret_analyze_cpp_impl [14.1s]

Summary: 1 package finished [14.4s]
```

takam5f2, 2022-10-06:
I'm concerned with PYTHONPATH after `source /path/to/caret/install/setup.bash`

Could you share the result of the following commands execution.

```
source /path/to/caret/install/setup.bash
echo $PYTHONPATH
```

Torreskai0722, 2022-10-06:
> 

I can find them, and there is no warning when building the caret_analyze_cpp_impl.

```
(base) hydrapc@hydrapc:~/projects/ros2_caret_ws$ find install/caret_analyze_cpp_impl/ | grep "\.so"
install/caret_analyze_cpp_impl/lib/libcaret_analyze_cpp_impl.so
install/caret_analyze_cpp_impl/lib/python3.8/site-packages/record_cpp_impl.cpython-37m-x86_64-linux-gnu.so
```
```
(base) hydrapc@hydrapc:~/projects/ros2_caret_ws$  colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-select caret_analyze_cpp_impl
Starting >>> caret_analyze_cpp_impl
Finished <<< caret_analyze_cpp_impl [0.41s]                     

Summary: 1 package finished [0.68s]

```

Torreskai0722, 2022-10-06:
> I can find them, and there is no warning when building the caret_analyze_cpp_impl.
>
> ```
> (base) hydrapc@hydrapc:~/projects/ros2_caret_ws$ find install/caret_analyze_cpp_impl/ | grep "\.so" [...]

> I'm concerned with PYTHONPATH after `source /path/to/caret/install/setup.bash`
> 
> Could you share the result of the following commands execution.
> 
> ```
> source /path/to/caret/install/setup.bash
> echo $PYTHONPATH
> ```

Here is the result:
```
(base) hydrapc@hydrapc:~/projects/ros2_caret_ws$ echo $PYTHONPATH
/home/hydrapc/projects/ros2_caret_ws/build/tracetools_launch:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_launch/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/ros2trace:/home/hydrapc/projects/ros2_caret_ws/install/ros2trace/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/tracetools_trace:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_trace/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/tracetools_read:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_read/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/ros2caret:/home/hydrapc/projects/ros2_caret_ws/install/ros2caret/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/install/caret_analyze_cpp_impl/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze:/home/hydrapc/projects/ros2_caret_ws/install/caret_analyze/lib/python3.8/site-packages:/home/hydrapc/projects/CARLA/CARLA_Leaderboard_20/PythonAPI/carla/:/home/hydrapc/projects/CARLA/scenario_runner:/home/hydrapc/projects/CARLA/leaderboard:/home/hydrapc/projects/CARLA/CARLA_Leaderboard_20/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg:/opt/ros/galactic/lib/python3.8/site-packages
```

Torreskai0722, 2022-10-06:
> I'm concerned with PYTHONPATH after `source /path/to/caret/install/setup.bash`
> 
> Could you share the result of the following commands execution.
> 
> ```
> source /path/to/caret/install/setup.bash
> echo $PYTHONPATH
> ```

Here is the result:
```
(base) hydrapc@hydrapc:~/projects/ros2_caret_ws$ echo $PYTHONPATH
/home/hydrapc/projects/ros2_caret_ws/build/tracetools_launch:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_launch/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/ros2trace:/home/hydrapc/projects/ros2_caret_ws/install/ros2trace/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/tracetools_trace:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_trace/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/tracetools_read:/home/hydrapc/projects/ros2_caret_ws/install/tracetools_read/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/ros2caret:/home/hydrapc/projects/ros2_caret_ws/install/ros2caret/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/install/caret_analyze_cpp_impl/lib/python3.8/site-packages:/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze:/home/hydrapc/projects/ros2_caret_ws/install/caret_analyze/lib/python3.8/site-packages:/home/hydrapc/projects/CARLA/CARLA_Leaderboard_20/PythonAPI/carla/:/home/hydrapc/projects/CARLA/scenario_runner:/home/hydrapc/projects/CARLA/leaderboard:/home/hydrapc/projects/CARLA/CARLA_Leaderboard_20/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg:/opt/ros/galactic/lib/python3.8/site-packages

```

takam5f2, 2022-10-11:
@Torreskai0722  
Sorry for not responding. Could you upload compressed log directory generated in `/path/to/caret/log`?

Best,

takam5f2, 2022-10-12:
@Torreskai0722  
Have you try building CARET just after deleting directories; `build` and `install`?
I suspect that cached data in `build` and `install` caused your issue.

If no, please try the following commands
```
cd /path/to/ros2_caret_ws
rm -rf build install log

colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Torreskai0722, 2022-10-12:
> @Torreskai0722 Have you tried building CARET just after deleting directories; `build` and `install`? I suspect that cached data in `build` and `install` caused your issue.
> 
> If no, please try the following commands
> 
> ```
> cd /path/to/ros2_caret_ws
> rm -rf build install log
> 
> colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
> ```

I tried to delete them and rebuint. The error is still there.

takam5f2, 2022-10-13:
@Torreskai0722  
Do you use Anaconda-based Python?  
I suppose that `(base)` on your terminal might imply that you are using Anaconda.
We have verified that CARET works on standard Python, but we haven't verified with Anaconda.

Torreskai0722, 2022-10-13:
> 

I deactivated the conda environment. Then I delete build, install and log folders and rebuild the caret. It works now! Thanks!
```
hydrapc@hydrapc:~/projects/ros2_caret_ws$ ros2 caret check_ctf -d ../ros2_ws/evaluate/e2e_sample/
Succeed to find record_cpp_impl. the C++ version will be used.
converting: 14143it [00:00, 35055.19it/s]                                       
14019 events found.
loading: 100%|████████████████████████| 14019/14019 [00:00<00:00, 146950.64it/s]
14019 events found.
loading: 100%|███████████████████████| 14019/14019 [00:00<00:00, 1518632.91it/s]
filtered to 222 events.
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):
/home/hydrapc/projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng_info.py:436: FutureWarning: In a future version of pandas, a length 1 tuple will be returned when iterating over a groupby with a grouper equal to a list of length 1. Don't supply a list with a single grouper to avoid this warning.
  for _, group_df in concat_df.groupby(['callback_group_addr']):

```

closed 2022-10-13 by Torreskai0722