tier4/caret #51: Failed in Autoware.universe compiling

Li-Jun-SD, 2022-11-10:
i compile autoware.universe with ros2 humble and caret in docker container. error occurs when some package compiling:
* euclidean_cluster
* image_projection_based_fusion
...
```
error: too few arguments to function ‘void ros_trace_rclcpp_publish(const void*, const void*, uint64_t)’
```
my command be like:
```
source /opt/ros/humble/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash
cd ~/autoware
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF --continue-on-error
```

keita1523, 2022-11-11:
@Li-Jun-SD 
autoware image is [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe)?
I want to prepare the same environments as you.

Li-Jun-SD, 2022-11-11:
> @Li-Jun-SD autoware image is [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe)? I want to prepare the same environments as you.

For some network reason, i could not pull those images directly. Ops engineer tell me images are based on those, just add user (use rocker? i am not sure.).
After run `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`, the compile is finished. Very confused.

keita1523, 2022-11-11:
Sorry.
The compile was finished, can I close this issue?

closed 2022-11-11 by Li-Jun-SD

takeshi-iwanari, 2022-11-25:
For someone who encounters the same error.

The latest (Oct 25~) Autoware (humble branch) uses [autowarefoundation/ament_cmake](https://github.com/autowarefoundation/ament_cmake/tree/autoware-humble) where SYSTEM is added to ament_auto_add_library. Due to this, CARET/rclcpp is not used for compilation and build error occurs. You can revert [this commit](https://github.com/autowarefoundation/ament_cmake/commit/6dc53769ada4c99dd1d061165b421fa11db0c769) by the following command. FYI, [this page](https://github.com/tier4/CARET_report/tree/main/sample_autoware) explains how to run Autoware with CARET.

```sh
cd ${autoware_dir}
cd src/core/external/ament_cmake
git revert -n 6dc53769ada4c99dd1d061165b421fa11db0c769
 ```