tier4/caret #52: How to define a target path by myself.

Li-Jun-SD, 2022-11-11:
Is it possible to define a target path by change architecture.yaml? I tried `paths = arch.search_paths()`, notebook busy forever..., no target_path is created.

Li-Jun-SD, 2022-11-11:
```
---------------------------------------------------------------------------
ItemNotFoundError                         Traceback (most recent call last)
/tmp/ipykernel_148960/2255407313.py in <module>
      1 from caret_analyze.plot import message_flow
      2 
----> 3 path = app.get_path('target_path')
      4 message_flow(path)

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/application.py in get_path(self, path_name)
    209             return x.path_name
    210 
--> 211         return Util.find_similar_one(path_name,
    212                                      self.paths,
    213                                      get_name)

~/ros2_caret_ws/build/caret_analyze/caret_analyze/common/util.py in find_similar_one(target_name, items, key, th)
    130             raise ItemNotFoundError(msg)
    131         else:
--> 132             raise ItemNotFoundError('Failed find item.')
    133 
    134     @staticmethod

ItemNotFoundError: Failed find item.
```

Li-Jun-SD, 2022-11-11:
```
named_paths:
- path_name: target_path
  node_chain:
  - node_name: /planning/mission_planning/mission_planner
    publish_topic_name: /planning/mission_planning/route
    subscribe_topic_name: UNDEFINED
  - node_name: /system/ad_service_state_monitor
    publish_topic_name: /autoware/state
    subscribe_topic_name: /planning/mission_planning/route
  - node_name: /system/system_error_monitor
    publish_topic_name: /system/emergency/hazard_status
    subscribe_topic_name: /autoware/state
  - node_name: /system/emergency_handler
    publish_topic_name: /system/emergency/hazard_lights_cmd
    subscribe_topic_name: /system/emergency/hazard_status
  - node_name: /control/vehicle_cmd_gate
    publish_topic_name: /control/command/turn_indicators_cmd
    subscribe_topic_name: /system/emergency/hazard_lights_cmd
  - node_name: /simulation/simple_planning_simulator
    publish_topic_name: /localization/kinematic_state
    subscribe_topic_name: /control/command/turn_indicators_cmd
  - node_name: /planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner
    publish_topic_name: UNDEFINED
    subscribe_topic_name: /localization/kinematic_state
```

takam5f2, 2022-11-11:
@Li-Jun-SD  

Please read [the page about `inter-node data path`](https://tier4.github.io/CARET_doc/latest/configuration/inter_node_data_path/).

> Is it possible to define a target path by change architecture.yaml? I tried paths = arch.search_paths(), notebook busy forever..., no target_path is created.

I think that `arch.search_paths()` tried to find many paths as possible, and it did not respond.  Please use `max_node_depth` or add another node as an argument to decrease possible candidates.

Best,

Li-Jun-SD, 2022-11-11:
> @Li-Jun-SD
>
> Please read [the page about `inter-node data path`](https://tier4.github.io/CARET_doc/latest/configuration/inter_node_data_path/).
>
>
> I think that `arch.search_paths()` tried to find many [...]

Thanks! 
Target_path is created after add `max_node_depth`. However, i could not get anything in Visulizaiton step. I run autoware planning simulation, and record it.
```
from caret_analyze.plot import message_flow

path = app.get_path('target_path')
path.verify()
```
```
WARNING : 2022-11-11 15:45:15 | Detected invalid message context. Correct these node_path definitions. 
To see node definition,execute [ app.get_node('/planning/scenario_planning/scenario_selector').summary.pprint() ] 
message_context: null
node: /planning/scenario_planning/scenario_selector
publish_topic_name: /planning/scenario_planning/scenario
subscribe_topic_name: /planning/mission_planning/route
```
```
False
```

takam5f2, 2022-11-11:
@Li-Jun-SD 

Please read the page about [intra-node data path](https://tier4.github.io/CARET_doc/latest/configuration/intra_node_data_path/).  I recommend you to add `message_context: use_latest_message` between `/planning/mission_planning/route` and `/planning/scenario_planning/scenario` in `/planning/scenario_planning/scenario_selector`. This requires you to edit the architecture file.

closed 2022-11-11 by Li-Jun-SD