tier4/caret #54: InvalidArgumentError: Contains an unknown columns. {'publisher_handle'}

Li-Jun-SD, 2022-11-24:
Hi,
I am rebuild autoware with caret and try to get some latency information.

```
source /opt/ros/humble/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash
source ~/autoware/install/local_setup.bash

export LD_PRELOAD=$(readlink -f ~/ros2_caret_ws/install/caret_trace/lib/libcaret.so)
# export CARET_IGNORE_NODES="/rviz*"
export CARET_IGNORE_TOPICS="/clock:/parameter_events"
export CARET_SELECT_NODES="/planning/mission_planning/mission_planner:/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner"
# export CARET_SELECT_TOPICS="/planning/mission_planning/route"

ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
```

Finally, I get a path like this:
```
paths = arch.search_paths(
'/planning/mission_planning/mission_planner', # source node
'/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner') # destination node

path:
- message_context: null
  node: /planning/mission_planning/mission_planner
- topic: /planning/mission_planning/route
- message_context: null
  node: /planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner
```

And `path.verify()` is true.

But when I want to visualization it, I got an error:
```
---------------------------------------------------------------------------
InvalidArgumentError                      Traceback (most recent call last)
/tmp/ipykernel_21223/2255407313.py in <module>
      2 
      3 path = app.get_path('target_path')
----> 4 message_flow(path)

~/ros2_caret_ws/build/caret_analyze/caret_analyze/plot/bokeh/message_flow.py in message_flow(path, export_path, granularity, treat_drop_as_delay, lstrip_s, rstrip_s, use_sim_time)
     73     color_palette = ColorPalette(Bokeh8)
     74 
---> 75     df = path.to_dataframe(treat_drop_as_delay=treat_drop_as_delay)
     76 
     77     converter: Optional[ClockConverter] = None

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path_base.py in to_dataframe(self, remove_dropped, treat_drop_as_delay, lstrip_s, rstrip_s, shaper)
    107 
    108         """
--> 109         records = self.to_records()
    110         column_names = self.column_names
    111 

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path_base.py in to_records(self)
     41 
     42         """
---> 43         return self.__records.clone()
     44 
     45     @abstractmethod

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path_base.py in __records(self)
     61     def __records(self) -> RecordsInterface:
     62         if self.__records_cache is None:
---> 63             self.__records_cache = self._to_records_core()
     64         return self.__records_cache
     65 

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path.py in _to_records_core(self)
    237     def _to_records_core(self) -> RecordsInterface:
    238         self._verify_path(self.node_paths)
--> 239         return RecordsMerged(self.child).data
    240 
    241     @staticmethod

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path.py in __init__(self, merge_targets)
    107         if len(merge_targets) == 0:
    108             raise InvalidArgumentError('There are no records to be merged.')
--> 109         self._data = self._merge_records(merge_targets)
    110 
    111     @property

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path.py in _merge_records(targets)
    123             targets = targets[1:]
    124 
--> 125         first_element = targets[0].to_records()
    126         left_records = first_element
    127 

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path_base.py in to_records(self)
     41 
     42         """
---> 43         return self.__records.clone()
     44 
     45     @abstractmethod

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/path_base.py in __records(self)
     61     def __records(self) -> RecordsInterface:
     62         if self.__records_cache is None:
---> 63             self.__records_cache = self._to_records_core()
     64         return self.__records_cache
     65 

~/ros2_caret_ws/build/caret_analyze/caret_analyze/runtime/communication.py in _to_records_core(self)
    267         """
    268         assert self._records_provider is not None
--> 269         records = self._records_provider.communication_records(self._val)
    270 
    271         return records

~/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/records_provider_lttng.py in communication_records(self, comm_val)
     93             return self._compose_intra_proc_comm_records(comm_val)
     94 
---> 95         return self._compose_inter_proc_comm_records(comm_val)
     96 
     97     def node_records(

~/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/records_provider_lttng.py in _compose_inter_proc_comm_records(self, comm_value)
    758         callback_object = self._helper.get_subscription_callback_object_inter(subscription_cb)
    759 
--> 760         records = self._source.inter_comm_records(publisher_handles, callback_object)
    761 
    762         columns = [COLUMN_NAME.RCLCPP_PUBLISH_TIMESTAMP]

~/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/records_provider_lttng.py in inter_comm_records(self, publisher_handles, callback_object)
   1429         drop = list(set(merged.columns) - set(columns))
   1430         merged.drop_columns(drop)
-> 1431         merged.reindex(columns)
   1432 
   1433         return merged

~/ros2_caret_ws/build/caret_analyze/caret_analyze/record/record_cpp_impl.py in reindex(self, columns)
    177             msg = 'Contains an unknown columns. '
    178             msg += f'{miss_match_columns}'
--> 179             raise InvalidArgumentError(msg)
    180         self._records.reindex(columns)
    181         self._columns.reindex(columns)

InvalidArgumentError: Contains an unknown columns. {'publisher_handle'}
```

I have no idea why this error occurs and what does it mean. Could you help me to fix it?

hsgwa, 2022-11-28:
Thanks for reporting the bug!
I hope that applying the following PR will solve the problem.
https://tier4.github.io/CARET_doc/latest/visualization/communication/pub_sub_information/applying PR.

However, there is a possibility that the latency cannot be calculated even if the error disappears after applying the PR.
It is also possible that CARET is not working properly or the topic name you are measuring is incorrect and so on.

- First, please confirm that the error disappears after applying the patch.
- After that, could you please check if publisher and subscription were working correctly?
Visualization methods are described here: 
https://tier4.github.io/CARET_doc/latest/visualization/communication/pub_sub_information/

Li-Jun-SD, 2022-11-29:
> /planning/mission_planning/mission_planner

I have checked publisher and subscriber. I use `export CARET_SELECT_TOPICS="/planning/scenario_planning/scenario_selector/trajectory", but here are still some topics i do not want record has been recorded.

```
 topic_name                                                                                                                          |   number_of_trace_points
-------------------------------------------------------------------------------------------------------------------------------------+--------------------------
 /localization/kinematic_state                                                                                                       |                    29586
 /parameter_events                                                                                                                   |                     6734
 /planning/scenario_planning/scenario_selector/trajectory                                                                            |                     2825
...
```

I use it after `export CARET_SELECT_NODES`.

`unknown columns` disappear, ` Failed to find any records that went through the path.There is a possibility that all records are lost.` now. And when I in configuration phase, `Tracer discarded 995 events` is always here. But  `ros2 caret topic_summary --trace_dir` show trace points exist.

Li-Jun-SD, 2022-11-29:
> Thanks for reporting the bug! I hope that applying the following PR will solve the problem. [tier4/CARET_analyze#210](https://github.com/tier4/CARET_analyze/pull/210)
> 
> ```
> cd ~/ros2_caret_ws/src/CARET_analyze/
> git fetch
> git checkout fix_unknown_columns
> ```
> 
> However, there is a possibility that the latency cannot be calculated even if the error disappears after applying the PR. It is also possible that CARET is not working properly or the topic name you are measuring is incorrect and so on.
> 
> * First, please confirm that the error disappears after applying the patch.
> * After that, could you please check if publisher and subscription were working correctly?
>   Visualization methods are described here:
>   https://tier4.github.io/CARET_doc/latest/visualization/communication/pub_sub_information/

I have checked publisher and subscriber. I use `export CARET_SELECT_TOPICS="/planning/scenario_planning/scenario_selector/trajectory", but here are still some topics i do not want record has been recorded.

```
 topic_name                                                                                                                          |   number_of_trace_points
-------------------------------------------------------------------------------------------------------------------------------------+--------------------------
 /localization/kinematic_state                                                                                                       |                    29586
 /parameter_events                                                                                                                   |                     6734
 /planning/scenario_planning/scenario_selector/trajectory                                                                            |                     2825
...
```

I use it after `export CARET_SELECT_NODES`.

`unknown columns` disappear, ` Failed to find any records that went through the path.There is a possibility that all records are lost.` now. And when I in configuration phase, `Tracer discarded 995 events` is always here. But  `ros2 caret topic_summary --trace_dir` show trace points exist.

hsgwa, 2022-11-30:
Thank you for the detailed report.

It seems that there is still an error in CARET_analyze if there is no measurement data.
> ValueError: min() arg is an empty sequence

I will report them as bugs.

Above all, we need to make adjustments to prevent `Tracer discarded`, because if even one lost event occurs, it cannot be measured correctly.

> Tracer discarded 995 events is always here

TOPIC_IGNORE and TOPIC_SELECT are currently not completely removed, but it reduces recording data.

The following is a count of the trace data in the demo code as a test.

```
export CARET_SELECT_TOPICS="/topic1" 
$ ros2 launch caret_demos  end_to_end_sample_with_lttng_session.launch.py

$ ros2 caret topic_summary -d ~/.ros/tracing/e2e_sample/
 topic_name        |   number_of_trace_points
-------------------+--------------------------
 /topic1           |                      103
 /parameter_events |                        7
 /rosout           |                        7
 /caret/status     |                        1
 /drive            |                        1
 /topic2           |                        1
 /topic3           |                        1
 /topic4           |                        1
```

There is less data for topics that are not selected, which suppresses `discarded`.

For reference, I share the IGNORE environment variables we use to measure Autoware.
https://github.com/takeshi-iwanari/autoware/blob/16fb26f365df64c4b7e279df35bdc41b72d7732b/caret_topic_filter.bash

Alternatively, it also suppresses `discarded` to increase the buffer size.
This requires manually fixing the source code.
https://github.com/tier4/ros2_tracing/pull/1/files

Li-Jun-SD, 2022-12-01:
Thanks for your help! It worked now. I got measurement data after rebuild rclcpp. Maybe I made some faults in building process. 
This tool really help me a lot, thanks again.

closed 2022-12-01 by Li-Jun-SD