tier4/caret #77: `ros2 caret check_ctf` failed after finishing recording

st9540808, 2023-02-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Recording

### Description

Hi, I am trying to use caret to trace CARET_demo as shown in the tutorial. I want to use caret within docker containers. However, I encountered the error from the command `ros2 caret check_ctf`

```
$ ros2 caret check_ctf
 -d ./ntu_ws/evaluate/e2e_sample/    
Succeed to find record_cpp_impl. the C++ version will be used.
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.5', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/st9540808/ros2_caret_ws/build/ros2caret/ros2caret/command/caret.py", line 32, in main
    return extension.main(args=args)
  File "/home/st9540808/ros2_caret_ws/build/ros2caret/ros2caret/verb/check_ctf.py", line 36, in main
    Lttng(args.trace_dir)
  File "/home/st9540808/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 283, in __init__
    data, events, begin, end = self._parse_lttng_data(
  File "/home/st9540808/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 314, in _parse_lttng_data
    event_collection = EventCollection(
  File "/home/st9540808/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 70, in __init__
    self._iterable_events = CtfEventCollection(trace_dir)
  File "/home/st9540808/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 205, in __init__
    assert begin_msg is not None
AssertionError
```

---

My environment is shown below:

| environment |  |
| --- | --- |
| OS | Ubutnu 22.04 |
| ROS 2 version | humble |
| caret version | v0.4.2 | 

As for the container, I build an image by instructions from [autoware documentation](https://github.com/autowarefoundation/autoware/tree/main/docker#building-docker-images-on-your-local-machine) but slightly modified it. I put the modified dockerfile on my [github](https://github.com/st9540808/ntu_ws/blob/master/docker/autoware-universe/Dockerfile). To simply put, I only disable sourcing ROS 2 `setup.bash` when building image. Therefore, it is nearly identical to the image from `ghcr.io/autowarefoundation/autoware-universe`. I run `setup-dev-env.sh` and build caret successfully with the image I build.

The steps I used are the same as described in the tutorial:
https://tier4.github.io/CARET_doc/latest/tutorials/recording/

I also confirmed that CARET_demos is built with caret-rclcpp
```
$ colcon build --symlink-install --packages-up-to caret_demos --cmake-args -DBUILD_TESTING=OFF
Starting >>> caret_demos
Finished <<< caret_demos [6.17s]                     

Summary: 1 package finished [6.28s]

$ ros2 caret check_caret_rclcpp --workspace .
INFO    : 2023-02-05 02:17:46 | All packages are built using caret-rclcpp.
```

Also, `ros2 caret record` seems to work correctly.
```
$ ros2 caret record -s e2e_sample
[INFO] [1675534724.286570434] [caret]: trace filtering is enabled.
[INFO] [1675534724.286585546] [caret]: CARET_IGNORE_NODES: /rviz*
[INFO] [1675534724.286587732] [caret]: CARET_IGNORE_TOPICS: /clock:/parameter_events
Error: No session daemon is available
[INFO] [1675534724.304793409] [caret_trace_112]: /caret_trace_112 started
UST tracing enabled (1 events)
kernel tracing disabled
context (3 names)
writing tracing session to: /home/st9540808/ntu_ws/evaluate/e2e_sample
press enter to start...
2 recordable processes found.
All process started recording.
press enter to stop...
stopping & destroying tracing session
```

I would suspect the error from `ros2 caret check_ctf` is due to LTTng not working in containers. I checked the evaluate directory and find out there is no file in `evaluate/e2e_sample/ust/uid/1000/64-bit/index`:

[image]

And the printable `metadata` content is shown below

```
/* CTF 1.8 */

typealias integer { size = 8; align = 8; signed = false; } := uint8_t;
typealias integer { size = 16; align = 8; signed = false; } := uint16_t;
typealias integer { size = 32; align = 8; signed = false; } := uint32_t;
typealias integer { size = 64; align = 8; signed = false; } := uint64_t;
typealias integer { size = 64; align = 8; signed = false; } := unsigned long;
typealias integer { size = 5; align = 1; signed = false; } := uint5_t;
typealias integer { size = 27; align = 1; signed = false; } := uint27_t;

trace {
	major = 1;
	minor = 8;
	uuid = "c66c74d6-961a-46c4-a5b0-31d1eeff1092";
	byte_order = le;
	packet.header := struct {
		uint32_t magic;
		uint8_t  uuid[16];
		uint32_t stream_id;
		uint64_t stream_instance_id;
	};
};

env {
	domain = "ust";
	tracer_name = "lttng-ust";
	tracer_major = 2;
	tracer_minor = 13;
	tracer_buffering_scheme = "uid";
	tracer_buffering_id = 1000;
	architecture_bit_width = 64;
	trace_name = "e2e_sample";
	trace_creation_datetime = "20230205T021847+0800";
	hostname = "st9540808-i9-13900K";
};

clock {
	name = "monotonic";
	uuid = "65c61372-d757-4682-a22b-d454baf7c7f4";
	description = "Monotonic Clock";
	freq = 1000000000; /* Frequency, in Hz */
	/* clock value offset from Epoch is: offset * (1/freq) */
	offset = 1675512980426787863;
};

typealias integer {
	size = 27; align = 1; signed = false;
	map = clock.monotonic.value;
} := uint27_clock_monotonic_t;

typealias integer {
	size = 32; align = 8; signed = false;
	map = clock.monotonic.value;
} := uint32_clock_monotonic_t;

typealias integer {
	size = 64; align = 8; signed = false;
	map = clock.monotonic.value;
} := uint64_clock_monotonic_t;

struct packet_context {
	uint64_clock_monotonic_t timestamp_begin;
	uint64_clock_monotonic_t timestamp_end;
	uint64_t content_size;
	uint64_t packet_size;
	uint64_t packet_seq_num;
	unsigned long events_discarded;
	uint32_t cpu_id;
};

struct event_header_compact {
	enum : uint5_t { compact = 0 ... 30, extended = 31 } id;
	variant <id> {
		struct {
			uint27_clock_monotonic_t timestamp;
		} compact;
		struct {
			uint32_t id;
			uint64_clock_monotonic_t timestamp;
		} extended;
	} v;
} align(8);

struct event_header_large {
	enum : uint16_t { compact = 0 ... 65534, extended = 65535 } id;
	variant <id> {
		struct {
			uint32_clock_monotonic_t timestamp;
		} compact;
		struct {
			uint32_t id;
			uint64_clock_monotonic_t timestamp;
		} extended;
	} v;
} align(8);
```

I would like to know the reason of this problem and some advice on protential LTTng problem. Thank you for your time!

takeshi-iwanari, 2023-02-06:
@st9540808 
Sorry for the inconvenience.
We enlarged buffer size for LTTng to avoid data lost: https://github.com/tier4/ros2_tracing/blob/humble_tracepoint_added/tracetools_trace/tracetools_trace/tools/lttng_impl.py#L157 . However, it causes no recorded trace data with some environments (on Docker and with many CPU cores). It happens only on Docker for some reasons...

Please decrease the number of the buffer size like the following.
`sed -i "s/256 \* 4096/32 \* 4096/" ~/ros2_caret_ws/src/ros-tracing/ros2_tracing/tracetools_trace/tracetools_trace/tools/lttng_impl.py`

st9540808, 2023-02-06:
@takeshi-iwanari 
Thank you for your quick reply! 
I will try it and check if the error still persists this week.

st9540808, 2023-02-15:
Hi @takeshi-iwanari, I am sorry to bother you again.

I can now generate LTTng data after following your instructions. The command `ros2 caret check_ctf` generates following output

```
$ ros2 caret check_ctf -d /home/st9540808/ntu_ws/evaluate/e2e_sample/
Succeed to find record_cpp_impl. the C++ version will be used.
75 events found.
loading: 100%|████| 75/75 [00:00<00:00, 145366.36it/s]
WARNING : 2023-02-15 16:09:25 | Failed to find trace point added by caret-rclcpp. To check whether binary are built with caret-rclcpp,run CARET CLI : ros2 caret check_caret_rclcpp.
75 events found.
loading: 100%|████| 75/75 [00:00<00:00, 342299.02it/s]
filtered to 62 events.
```

However, I encountered another problem when doing the tutorial at https://tier4.github.io/CARET_doc/main/tutorials/visualization/. The error is shown below:

[image]

I guess the error is related to the warning message `Failed to find trace point added by caret-rclcpp`. However, I check the CARET_demo workspace using `ros2 caret check_caret_rclcpp`, and it said all packages are built using caret-rclcpp. Also, I check if the CARET_demo binary is linked with caret shared binary. It seems that it linked to the correct libraries.

```
$ ldd /home/st9540808/ntu_ws/CARET_demos/build/caret_demos/end_to_end_sample | grep caret
        /home/st9540808/ros2_caret_ws/install/caret_trace/lib/libcaret.so (0x00007fc7e0587000)
        librclcpp.so => /home/st9540808/ros2_caret_ws/install/rclcpp/lib/librclcpp.so (0x00007fc7e02e9000)
        librcl.so => /home/st9540808/ros2_caret_ws/install/rcl/lib/librcl.so (0x00007fc7e02a0000)
        libtracetools.so => /home/st9540808/ros2_caret_ws/install/tracetools/lib/libtracetools.so (0x00007fc7e0283000)
        libcaret_msgs__rosidl_typesupport_cpp.so => /home/st9540808/ros2_caret_ws/install/caret_msgs/lib/libcaret_msgs__rosidl_typesupport_cpp.so (0x00007fc7dfcd4000)
        librcl_yaml_param_parser.so => /home/st9540808/ros2_caret_ws/install/rcl_yaml_param_parser/lib/librcl_yaml_param_parser.so (0x00007fc7df455000)
```

I have no clue why the error happened. Thank you again for your time in looking into my problem.

st9540808, 2023-02-15:
I just found out the solution by myself.

It seems that `caret record` and `end_to_end_sample.launch.py` need to run in the same container, otherwise the profiled application cannot see the LTTng session daemon.

I can now run the python that I showed in the screenshot above. I will try to use visualization next.

closed 2023-02-15 by st9540808