tier4/caret #93: ubuntu20.04 humble caret

Autostone-c, 2023-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Build

### Description

If I want to use caret in the ROS2 (Humble) version on Ubuntu 20.04, I cannot use setup_caret. sh installation related dependencies?

Autostone-c, 2023-08-03:
> 如果您渴望在 ROS 2 Humble + Ubuntu 20.04 上使用 CARET，请尝试手动安装，而不是`./setup_caret.bash` https://tier4.github.io/CARET_doc/latest/installation/manual_installation/

hi, i try manual_installation on ubuntu20.04 and humble branch ,but i have the error which cant sovle
[image]

takeshi-iwanari, 2023-08-08:
@Autostone-c 
Thank you for sharing the result.
The error you encountered is our bad. We recently added [a new code](https://github.com/tier4/ros2caret/pull/69/files#diff-60f61ab7a8d1910d86d9fda2261620314edcae5894d5aaa236b821c7256badd7R7) which doesn't work before Python 3.10.

I uploaded a change to fix it. Please run the following commands, then run `colcon` again. Building CARET succeeded in my environment.

```
cd src/CARET/ros2caret
git pull
git checkout fix_annotation
cd ../../../
```

By the way, CARET needs the following packages, but install will fail on Humble on Ubuntu 20.04 ( `E: Unable to locate package ros-humble-ros2trace` ). How did you install them? (or did you manually build them?)

```
apt install -y \
  ros-humble-ros2trace \
  ros-humble-ros2trace-analysis \
  ros-humble-tracetools \
  ros-humble-tracetools-analysis \
  ros-humble-tracetools-launch \
  ros-humble-tracetools-read \
  ros-humble-tracetools-test \
  ros-humble-tracetools-trace
```

Autostone-c, 2023-08-08:
> @Autostone-c Thank you for sharing the result.
> 
> ## Solution for the error
> The error you encountered is our bad. We recently added [a new code](https://github.com/tier4/ros2caret/pull/69/files#diff-60f61ab7a8d1910d86d9fda2261620314edcae5894d5aaa236b821c7256badd7R7) which doesn't work before Python 3.10.
> 
> I uploaded a change to fix it. Please run the following commands, then run `colcon` again. Building CARET succeeded in my environment.
> 
> ```shell
> cd src/CARET/ros2caret
> git pull
> git checkout fix_annotation
> cd ../../../
> ```
> 
> ## Question about installing packages
> By the way, CARET needs the following packages, but install will fail on Humble on Ubuntu 20.04 ( `E: Unable to locate package ros-humble-ros2trace` ). How did you install them? (or did you manually build them?)
> 
> ```shell
> apt install -y \
>   ros-humble-ros2trace \
>   ros-humble-ros2trace-analysis \
>   ros-humble-tracetools \
>   ros-humble-tracetools-analysis \
>   ros-humble-tracetools-launch \
>   ros-humble-tracetools-read \
>   ros-humble-tracetools-test \
>   ros-humble-tracetools-trace
> ```
> 
> ## FYI: My environment
> ```shell
> docker pull ubuntu:20.04
> docker run -it --rm ubuntu:20.04
> ```
> 
> The following commands are executed in Docker (Ubuntu 20.04), so `sudo` is removed.
> 
> ```shell
> apt update
> 
> # Install Python 3.10 (ROS 2 build will fail with this)
> # apt install build-essential zlib1g-dev libncurses5-dev libgdbm-dev libnss3-dev libssl-dev libreadline-dev libffi-dev libsqlite3-dev wget libbz2-dev
> # wget https://www.python.org/ftp/python/3.10.3/Python-3.10.3.tgz
> # tar -xzvf Python-3.10.3.tgz
> # cd Python-3.10.3/
> # ./configure --enable-optimizations
> # make -j 8
> # make altinstall
> # update-alternatives --install /usr/bin/python python /usr/local/bin/python3.10 1
> # update-alternatives --install /usr/bin/python3 python3 /usr/local/bin/python3.10 1
> # echo "alias pip=pip3.10" >> ~/.bashrc
> # echo "alias pip3=pip3.10" >> ~/.bashrc
> # source ~/.bashrc
> 
> 
> # Install ROS2 https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html#system-setup
> locale  # check for UTF-8
> apt update && apt install locales
> locale-gen en_US en_US.UTF-8
> update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
> export LANG=en_US.UTF-8
> locale  # verify settings
> 
> apt install software-properties-common
> add-apt-repository universe
> 
> apt update && apt install curl -y
> curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
> 
> echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
> 
> apt update && apt install -y \
>   python3-flake8-docstrings \
>   python3-pip \
>   python3-pytest-cov \
>   ros-dev-tools
> 
> python3 -m pip install -U \
>    flake8-blind-except \
>    flake8-builtins \
>    flake8-class-newline \
>    flake8-comprehensions \
>    flake8-deprecated \
>    flake8-import-order \
>    flake8-quotes \
>    "pytest>=5.3" \
>    pytest-repeat \
>    pytest-rerunfailures
> 
> mkdir -p ~/ros2_humble/src
> cd ~/ros2_humble
> vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src
> apt upgrade
> rosdep init
> rosdep update
> rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"
> colcon build --symlink-install
> 
> # . ~/ros2_humble/install/local_setup.bash
> # ros2 run demo_nodes_cpp talker
> 
> # Install CARET
> apt-add-repository ppa:lttng/stable-2.13
> apt update
> apt install lttng-tools liblttng-ust-dev
> apt install python3-babeltrace python3-lttng
> 
> apt update && apt install -y \
>   build-essential \
>   cmake \
>   git \
>   python3-colcon-common-extensions \
>   python3-flake8 \
>   python3-pip \
>   python3-pytest-cov \
>   python3-rosdep \
>   python3-setuptools \
>   python3-vcstool \
>   wget
>   # python3-bt2 \
> 
> python3 -m pip install -U \
>   flake8-blind-except \
>   flake8-builtins \
>   flake8-class-newline \
>   flake8-comprehensions \
>   flake8-deprecated \
>   flake8-docstrings \
>   flake8-import-order \
>   flake8-quotes \
>   pytest-repeat \
>   pytest-rerunfailures \
>   pytest \
>   setuptools \
>   colorcet
> 
> # apt install -y \
> #   ros-humble-ros2trace \
> #   ros-humble-ros2trace-analysis \
> #   ros-humble-tracetools \
> #   ros-humble-tracetools-analysis \
> #   ros-humble-tracetools-launch \
> #   ros-humble-tracetools-read \
> #   ros-humble-tracetools-test \
> #   ros-humble-tracetools-trace
> 
> . ~/ros2_humble/install/local_setup.bash
> 
> git clone https://github.com/tier4/caret.git ros2_caret_ws
> cd ros2_caret_ws
> mkdir src
> vcs import src < caret.repos
> rosdep install \
>   --from-paths src --ignore-src \
>   --rosdistro humble -y \
>   --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"
> 
> ln -sf ~/ros2_caret_ws/src/ros-tracing/ros2_tracing/tracetools/include/tracetools ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/
> ln -sf ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp_action/include/rclcpp_action ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/
> ln -sf ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp_components/include/rclcpp_components/ ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/
> ln -sf ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp_lifecycle/include/rclcpp_lifecycle/ ~/ros2_caret_ws/src/ros2/rclcpp/rclcpp/include/
> 
> colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=off  --symlink-install
> ```

i will try it. yep, i build all manually

Autostone-c, 2023-08-08:
> i will try it. yep, i build all manually

it works. but when i run (ros2 caret check_ctf) it occur error
[image]

takeshi-iwanari, 2023-08-08:
@Autostone-c 
CARET_analyze module, which is called from `ros2 caret check_ctf` command, has dependency on Python 3.10. Therefore, it doesn't work with Python 3.10 as described in [Requirements](https://tier4.github.io/CARET_doc/main/installation/installation/#requirements).

My recommendation is,

1. Record your target application using CARET on Humble + Ubuntu 20.04 (the environment you manually built)
2. Then, copy the recorded data by step 1 to another PC (Humble + Ubuntu 22.04) and analyze it  (e.g. call `ros2 caret check_ctf` , create an analysis script using CARET_analyze, etc.) 

Note: Again, CARET doesn't support Humble+Ubuntu20.04 nor Python 3.8. So, we are not sure the recording process (step 1) is successful.

[image]

Autostone-c, 2023-08-08:
> @Autostone-c CARET_analyze module, which is called from `ros2 caret check_ctf` command, has dependency on Python 3.10. Therefore, it doesn't work with Python 3.10 as described in [...]

thanks for your suggestion. i will try

Autostone-c, 2023-08-09:
> @Autostone-c CARET_analyze module, which is called from `ros2 caret check_ctf` command, has dependency on Python 3.10. Therefore, it doesn't work with Python 3.10 as described in [...]

hi, i use 2 to get a graph.is that right?
[image]

takeshi-iwanari, 2023-08-09:
The message flow graph you uploaded looks wrong. It should be like the following diagram (recorded in Humble + Ubuntu 20.04, and visualized in Humble + Ubuntu 22.04)

[image]

- I attach a jupyter note book. Could you try with this? (Please replace `tracing_log_path` )
- Could you share your trace data? (file in `~/.ros/tracing/` )

Autostone-c, 2023-08-09:
> The message flow graph you uploaded looks wrong. It should be like the following diagram (recorded in Humble + Ubuntu 20.04, and visualized in Humble + Ubuntu 22.04)
>
> [image]
>
> * I attach [a jupyter [...]

[evaluate.tar.gz](https://github.com/tier4/caret/files/12300941/evaluate.tar.gz)
this is my caret_demo trace data

takeshi-iwanari, 2023-08-09:
I checked your trace data and found an important trace point, named `ros2_caret:dds_bind_addr_to_stamp` , is lacked.
I guess you built cyclonedds by yourself. As a known issue, we found cyclonedds 0.10.x doesn't support the tracepoint.
If you use cyclonedds 0.10.x, please try either of following:

- Use cyclonedds0.9.x
- Apply this patch: https://github.com/eclipse-cyclonedds/cyclonedds/commit/bcdaf065083da970b5a976bf29fd76bb68e943ac

Autostone-c, 2023-08-09:
> I checked your trace data and found an important trace point, named `ros2_caret:dds_bind_addr_to_stamp` , is lacked. I guess you built cyclonedds by yourself. According to our investigation, cyclonedds 0.10.x doesn't support the trace point due to this [PR](https://github.com/eclipse-cyclonedds/cyclonedds/commit/bcdaf065083da970b5a976bf29fd76bb68e943ac). If you use cyclonedds 0.10.x, please try either of followings:
> 
> * Use fastdds instead of cyclonedds
> * Use cyclonedds0.9.x
> * Remove [this line](https://github.com/eclipse-cyclonedds/cyclonedds/blob/fe899f265dfa42d6303ef81487e4dfe8b9e203c1/src/core/CMakeLists.txt#L20), when you build cyclonedds 0.10.x

really really thanks , it solve my error

closed 2023-08-09 by Autostone-c