tier4/caret #104: check_ctf error

Autostone-c, 2023-09-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Visualization

### Description

when i run (ros2 caret check_ctf -d ~/ros2_ws/evaluate/e2e_sample/) i meet error
[image]

takeshi-iwanari, 2023-09-04:
Please install pandas < 2.1.0
See https://github.com/tier4/CARET_analyze/pull/324 for more detail.

Autostone-c, 2023-09-05:
> Please install pandas < 2.1.0 See [tier4/CARET_analyze#324](https://github.com/tier4/CARET_analyze/pull/324) for more detail.

thanks for this help. 
and i also want to ask if i can move all caret.repos packages to humble（replace rclcpp ros2_tracing and add the othes packages） and compile by manually。then i only source humble/install/setup.bash to recording？

takeshi-iwanari, 2023-09-05:
> and i also want to ask if i can move all caret.repos packages to humble（replace rclcpp ros2_tracing and add the othes packages） and compile by manually。then i only source humble/install/setup.bash to recording？

I didn't try it but it seems a good idea! I think it's worth a try.

Autostone-c, 2023-09-06:
> > and i also want to ask if i can move all caret.repos packages to humble（replace rclcpp ros2_tracing and add the othes packages） and compile by manually。then i only source humble/install/setup.bash to recording？
> 
> I didn't try it but it seems a good idea! I think it's worth a try.

thanks，i will try it. and close this issues

closed 2023-09-06 by Autostone-c