tier4/caret #110: Simple Dockerfile installation fails

Tobias-Pe, 2023-09-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Help wanted

### Step where the issue is related to

Build

### Environments

- OS: Windows with WSL (Ubuntu 22.04) 
- Docker setup:

### CARET version

Caret: 0.4.15 & 0.4.10 tried out

### Description

My Dockerfile, which is super close to your dockerfile:

```Docker
FROM osrf/ros:humble-desktop

RUN apt-get update -y && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends \
        locales=2.35-0ubuntu3.1 \
        && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*
RUN locale-gen en_US.UTF-8
ENV LANG en_US.UTF-8

ARG CARET_VERSION="v0.4.15"
ENV WORKSPACE_CARET=/workspaces/ros2_caret_ws

RUN git clone https://github.com/tier4/caret.git ${WORKSPACE_CARET} && \
    cd ${WORKSPACE_CARET} && \
    git checkout "$CARET_VERSION"

RUN cd ${WORKSPACE_CARET} && \
    rm -rf src build log install && \
    mkdir src && \
    vcs import src < caret.repos && \
    . /opt/ros/"$ROS_DISTRO"/setup.sh && \
    ./setup_caret.sh -c

RUN cd ${WORKSPACE_CARET} && \
    . /opt/ros/"$ROS_DISTRO"/setup.sh && \
    colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF

```

But it fails:

[docker_build.log](https://github.com/tier4/caret/files/12570179/docker_build.log)

Smth isnt really compatible here :/ and i cant figure it out at all.

### Steps to reproduce

copy my snippet into a Dockerfile

docker build -f Dockerfile .

### Commands

```shell
docker build -f Dockerfile .
```

### Logs

```shell
[docker_build.log](https://github.com/tier4/caret/files/12570179/docker_build.log)
```

nabetetsu, 2023-09-11:
This error is expected to resolved in this PR: (https://github.com/tier4/CARET_analyze/pull/330)
The cause is due to the latest version of `setuptools` that CARET references, this error is avoided by dropping the `setuptools` version as a workaround.

Tobias-Pe, 2023-09-11:
the workaround did the job 🎉 

> This error is expected to resolved in this PR: ([tier4/CARET_analyze#330](https://github.com/tier4/CARET_analyze/pull/330)) The cause is due to the latest version of `setuptools` that CARET references, this error is avoided by dropping the `setuptools` version as a workaround.

closed 2023-09-11 by Tobias-Pe