tier4/caret #135: Cannot install caret on Ubuntu 20.04 with ROS2 Galactic

ZiweiSong96, 2023-09-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Help wanted

### Step where the issue is related to

Build

### Environments

OS: Ubuntu 20.04.5 LTS
Kernel version: Linux 5.15.0-84-generic
Architecture: X86-64
ROS2: Galactic

### CARET version

1:
CARET: galactic
CARET_analyze: main
CARET_trace: v0.2.3
CARET_analyze_cpp_impl: main
CARET/caret_common: main

2:
CARET: v0.2.3
CARET_analyze: v0.2.3
CARET_trace: v0.2.3
CARET/CARET_analyze_cpp_impl: v0.2.3

### Target application

Application name: Autoware.Universe

### Description

I build the Autoware.Universe on Ubuntu 20.04 with ROS2 Galactic. The environment of my system was build by setup-dev-env.sh ( the environment bash provided by Autoware.Universe). All the functions of Autoware.Universe are correct.

When I try to build the CARET tools for tracing, I met some questions and bugs; At first I followed the documents on https://tier4.github.io/CARET_doc/refs-tags-v0.2.3/ since the documents of main branch says if I want to build CARET on Ubuntu 20.04 with ROS2 Galactic, I should follow the Installation Documents of version v0.2.3. So I clone the git repos of CARET and execute "git checkout galactic"; I followed the steps and build&compile successfully, when I validated the trace data of CARET demos (also the galactic branch) by executing "ros2 caret check_ctf -d ~/ros2_ws/evaluate/e2e_sample/", I met the KeyError as follows:
  from pkg_resources import load_entry_point
Failed to load entry point 'check_ctf': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
Failed to load entry point 'create_architecture_file': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
Failed to load entry point 'node_summary': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
Failed to load entry point 'topic_summary': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
Failed to load entry point 'trace_point_summary': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
Failed to load entry point 'verify_paths': unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'
usage: ros2 caret [-h] Call `ros2 caret <command> -h` for more detailed usage. ...

optional arguments:
  -h, --help            show this help message and exit

Commands:
  check_caret_rclcpp  
  record              

  Call `ros2 caret <command> -h` for more detailed usage.

TypeError: unsupported operand type(s) for |: 'ABCMeta' and 'ABCMeta'

I thought that was because the caret repos were wrong, so I changed the CARET version to v0.2.3, and followed the installation steps, then I met the following issues:

[image]
[image]
[image]

### Steps to reproduce

1:
git clone https://github.com/tier4/caret.git ros2_caret_ws
cd ros2_caret_ws
git checkout galactic
mkdir src
vcs import src < caret.repos
./setup_caret.sh
source /opt/ros/galactic/setup.bash
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
source ~/ros2_caret_ws/install/local_setup.bash
ros2 run tracetools status

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
git clone https://github.com/tier4/CARET_demos.git src/CARET_demos --recursive
cd src/CARET_demos
git checkout galactic
cd ../../
source /opt/ros/galactic/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash # please keep the order after 'source /opt/ros/galactic/setup.bash'
colcon build --symlink-install --packages-up-to caret_demos --cmake-args -DBUILD_TESTING=OFF
ros2 caret check_caret_rclcpp --workspace ~/ros2_ws/
source /opt/ros/humble/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash
source ~/ros2_ws/install/local_setup.bash

export LD_PRELOAD=$(readlink -f ~/ros2_caret_ws/install/caret_trace/lib/libcaret.so)
ros2 launch caret_demos end_to_end_sample.launch.py
Open a new terminal:
source /opt/ros/galactic/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash
mkdir -p ~/ros2_ws/evaluate
export ROS_TRACE_DIR=~/ros2_ws/evaluate
ros2 caret record -s e2e_sample
ros2 caret check_ctf -d ~/ros2_ws/evaluate/e2e_sample/

2: 
git clone https://github.com/tier4/caret.git ros2_caret_ws
cd ros2_caret_ws
git checkout v0.2.3
mkdir src
vcs import src < caret.repos
./setup_caret.sh
source /opt/ros/galactic/setup.bash
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

nabetetsu, 2023-09-28:
Hi, @ZiweiSong96 , 
Please try this workaround for build errors.  https://tier4.github.io/CARET_doc/main/faq/known_issues/#build-using-ament_cmake

OsamaThakur, 2023-10-04:
Hi all, Facing the same issue, @nabetetsu. The build-using-ament_cmake workaround didnt help

nabetetsu, 2023-10-04:
Hi, @OsamaThakur , I'm sorry for the inconvenience.
Have you also built CARET with Ubuntu 20.04 + galactic environment?
In practice, working with galactic is difficult to support as it is treated as optional.
Please try it in Ubuntu 22.04 + humble environment as described in the requrements.
https://tier4.github.io/CARET_doc/main/installation/installation/

ZiweiSong96, 2023-10-16:
Thanks a lot for the help, I changed my system to Ubuntu 22.04, Ubuntu 20.04 with ROS2 galactic has lot of bugs

ymski, 2024-06-28:
I am closing this issue as the problem appears to be resolved.

closed 2024-06-28 by ymski