tier4/caret #154: Can CARET work across multiple machines?

justin-kel, 2023-10-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Recording

### Description

Hi I would like to check if it is possible for CARET to work across multiple remote machines.

I have tried to run a talker and and listener from the demo_nodes_cpp package, and then attempt to trace communication latency across machines in the same wireless LAN. However running ros2 caret record only sees the node running on my local machine. Is this expected behaviour, or is something likely wrong with my configuration?

miyakoshi-dev, 2023-10-11:
Thank you for your inquiry
caret currently can only be recorded on the local machine. It is planned that it will be possible to record between remote machines in the near future, but the date of support has not yet been determined.

justin-kel, 2023-10-11:
Hi noted, thank you very much for your response.

closed 2023-10-11 by justin-kel