tier4/caret #158: IndexError: list index out of range

OsamaThakur, 2023-10-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Help wanted

### Step where the issue is related to

Visualization

### Environments

- ENV - DOCKER
- OS: ros:humble-ros-base | Ubuntu 22.04.1 LTS
- Kernel version: 5.15.0-86-generic
- ROS 2: Humble
- CPU: 11th Gen Intel® Core™ i7-11850H @ 2.50GHz × 16 
- Memory: 31.0 GiB

### CARET version

- CARET: main | 70c1cce76c12cfe5e2806b267f6e5deadd72a479
- CARET_analyze: main | 6dd09f3bbe8e1c904e955d8d2328beccb91f9279
- CARET_trace: main | c80b9525fb84d74b564a6d124289ab61d3316ee2

### Target application

- Application - Propriety AD Stack 

### Description

Hi Team, I run into an issue with a specific node(Planner). I have attached the image for reference, I would like to know how to approach the debugging . Also the doc website seems to be down

### Steps to reproduce

.

### Screenshots

[image]

Along with this warning
WARNING : 2023-10-12 09:24:29 | Multiple pieces of values matching the search condition are found. callback_name: /vehicle0/planner/callback_02
callback_type:
  type_name: subscription_callback
callback_type_name: subscription_callback
construction_order: 0
node_name: /vehicle0/planner
publish_topic_names: null
service_name: null
subscribe_topic_name: /vehicle0/VEH/ax
summary: !!python/object:caret_analyze.common.summary.Summary
  data:
    name: /vehicle0/planner/callback_02
    node: /vehicle0/planner
    topic: /vehicle0/VEH/ax
    type: subscription_callback

ymski, 2023-10-13:
Hi @OsamaThakur . 
Looking at the attached image, there are several possible reasons for the error.
First, it would be a good idea to check that you have the necessary data to draw the graph.
In caret, there is a way to check the measured data using a pandas dataframe as shown below.
```python
target_node = '/vehicle0/planner'
target_idx = 0
node = app.get_node(target_node)
node.callbacks[target_idx].to_dataframe()
```

We may be able to provide more detailed advice if you share your analysis scripts, architecture files, and trace data.

OsamaThakur, 2023-10-13:
Hi @ymski , Thanks for getting back to me, Your tips helped me identify some problems, Could I have your email please so I can share the requested files

ymski, 2023-10-16:
Hi @OsamaThakur .
Is it possible to zip the file and attach it to Github?
If the above method is difficult, we will consider another method.

OsamaThakur, 2023-10-16:
Hey @ymski, I would like to share those files privately, So what could be the another method?

ymski, 2023-10-16:
hi, @OsamaThakur .
We understand that you want to share the data confidentially.
Then please send the data to yamasaki(`at`)isp.co.jp
(`at` should be replaced.)

ymski, 2023-10-30:
@OsamaThakur 
I apologize for the delay in my response. 
After checking the data you provided, I have identified the following:

- Tracedata:  `e2e_demo003`
  - This trace data is missing the necessary event "add_callback_group."
  - "add_callback_group" is an event recorded during initialization.
  - The error occurs because essential data required to understand the application's node configuration is missing.

- Tracedata: `e2e_demo003_25`
  - The error observed when using this trace data is due to a bug on the CARET. 
    - There doesn't seem to be any data defects similar to Trace Data A.
  - A fix for this issue is planned in the future.

OsamaThakur, 2023-11-06:
Thanks for getting back to me and for your inputs @ymski, feel free to close this ticket if you think, its not necessary for this issue to be open

ymski, 2024-06-25:
@OsamaThakur 
I apologize for the very long delay in contacting you. The following PR has enabled us to analyze the trace data you provided. Thank you very much for providing useful information to fix the bug in CARET.
- https://github.com/tier4/caret_analyze/pull/460

closed 2024-06-25 by ymski