tier4/caret #169: How to complie autoware.universe with CARET?

ruoxianglee, 2023-10-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Build

### Description

I am building Autoware.Universe project with CARET tool to trace some information for this app as in the paper of CARET. I have succefully executed the caret_demos packages in tutorial. And I follow the [tutorial](https://tier4.github.io/CARET_doc/main/tutorials/) and the source installation of Autoware.Universe in the [document](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/). The following are the instructions to build Autoware:
```
source /opt/ros/humble/setup.bash
source ~/ros2_caret_ws/install/local_setup.bash
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
colcon build --symlink-install --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release
```

But, it always failed. Some warnings for several packages as follows:
```
Cannot generate a safe runtime search path for target xxx because there is a cycle in the constraint graph.
```
And it finally aborted the building at ekf_localizer package with the following error:
```
/usr/bin/ld: libekf_localizer_lib.so: undefined reference to `ros_trace_message_construct'
/usr/bin/ld: libekf_localizer_lib.so: undefined reference to `ros_trace_dispatch_intra_process_subscription_callback'
/usr/bin/ld: libekf_localizer_lib.so: undefined reference to `ros_trace_rclcpp_intra_publish'
/usr/bin/ld: libekf_localizer_lib.so: undefined reference to `ros_trace_dispatch_subscription_callback'
```

I think it may be not an issue but some operation faults by myself when building it. So I opened this quetion to require some help. 

Thanks in advance.

xygyo77, 2023-10-23:
These are known issues and you need to apply the following patch to your ros2_humble system.
[https://tier4.github.io/caret_doc/main/faq/known_issues/#build](url)
Can you please try it?

xygyo77, 2023-10-23:
These are known issues and you need to apply the following patch to your ros2_humble system.  
[https://tier4.github.io/caret_doc/main/faq/known_issues/#build](https://tier4.github.io/caret_doc/main/faq/known_issues/#build)  

Can you please try it?

ruoxianglee, 2023-10-24:
Thanks for your response.

I tried. The question mentioned above was solved.

However, I just encountered another issue. The build log is shown as follows:

```

In file included from /usr/include/pcl-1.12/pcl/point_types.h:354,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/planner_data.hpp:41,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/scene_module_interface.hpp:18,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/src/scene_module_interface.cpp:15:
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:241:12: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  241 |     struct { \
      |            ^
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:259:3: note: in expansion of macro ‘PCL_ADD_UNION_POINT4D’
  259 |   PCL_ADD_UNION_POINT4D \
      |   ^~~~~~~~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:337:5: note: in expansion of macro ‘PCL_ADD_POINT4D’
  337 |     PCL_ADD_POINT4D; // This adds the members x,y,z which can also be accessed using the point (which is float[4])
      |     ^~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:289:7: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  289 |       { \
      |       ^
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:313:3: note: in expansion of macro ‘PCL_ADD_UNION_RGB’
  313 |   PCL_ADD_UNION_RGB \
      |   ^~~~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:368:5: note: in expansion of macro ‘PCL_ADD_RGB’
  368 |     PCL_ADD_RGB;
      |     ^~~~~~~~~~~
In file included from /usr/include/pcl-1.12/pcl/point_types.h:354,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/planner_data.hpp:41,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/scene_module_interface.hpp:18,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/src/scene_module_interface.cpp:15:
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:318:5: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  318 |     { \
      |     ^
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:411:5: note: in expansion of macro ‘PCL_ADD_INTENSITY’
  411 |     PCL_ADD_INTENSITY;
      |     ^~~~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:411:22: error: extra ‘;’ [-Werror=pedantic]
  411 |     PCL_ADD_INTENSITY;
      |                      ^
      |                      -
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:324:5: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  324 |     { \
      |     ^
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:437:5: note: in expansion of macro ‘PCL_ADD_INTENSITY_8U’
  437 |     PCL_ADD_INTENSITY_8U;
      |     ^~~~~~~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:437:25: error: extra ‘;’ [-Werror=pedantic]
  437 |     PCL_ADD_INTENSITY_8U;
      |                         ^
      |                         -
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:330:5: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  330 |     { \
      |     ^
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:469:5: note: in expansion of macro ‘PCL_ADD_INTENSITY_32U’
  469 |     PCL_ADD_INTENSITY_32U;
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:469:26: error: extra ‘;’ [-Werror=pedantic]
  469 |     PCL_ADD_INTENSITY_32U;
      |                          ^
      |                          -
In file included from /usr/include/pcl-1.12/pcl/point_types.h:354,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/planner_data.hpp:41,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/scene_module_interface.hpp:18,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/src/scene_module_interface.cpp:15:
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:241:12: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  241 |     struct { \
[... 388 lines cut ...]
      |       ^
In file included from /usr/include/pcl-1.12/pcl/point_types.h:354,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/planner_data.hpp:41,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/include/behavior_velocity_planner_common/scene_module_interface.hpp:18,
                 from /home/cpsiot/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner_common/src/scene_module_interface.cpp:15:
/usr/include/pcl-1.12/pcl/impl/point_types.hpp:241:12: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
  241 |     struct { \
      |            ^
[... cut ...]
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/behavior_velocity_planner_common.dir/build.make:76: CMakeFiles/behavior_velocity_planner_common.dir/src/scene_module_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/behavior_velocity_planner_common.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< behavior_velocity_planner_common [1min 35s, exited with code 2]
``` 

I am still working on this issue but have not fixed it yet. 

Thank you so much.

ruoxianglee, 2023-10-24:
The above issue (i.e., the errors when building behavior_velocity_planner_common) can be fixed by following:
[Autoware issue 1219](https://github.com/autowarefoundation/autoware.universe/issues/1219).

Thanks.

closed 2023-10-26 by ymski