tier4/caret #177: DispatchError from the multimethod package when validating traces

m-alcon, 2023-10-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Help wanted

### Step where the issue is related to

Visualization

### Environments

- **OS**: Ubuntu 22.04.3 LTS
- **Kernel version**: 6.2.0-35-generic
- **ROS**: Humble
- **CPU**: 12th Gen Intel(R) Core(TM) i7-1270P x 16
- **Memory**: 16.0 GiB
- **Python version**: 3.12.10

### CARET version

- **CARET**: main | [cd04e9b](https://github.com/tier4/caret/commit/cd04e9be546210e28ed20816dcbdc6b6fc030bff)
- **CARET_analyze**: 5f608de07f84c9fb2dea23d8a360a91acbccfcbd
- **CARET_trace**: 5c0b803b936e64f2b6d1e2519796036b425f8098

### Target application

- Application name: Autoware
- URL: https://github.com/autowarefoundation/autoware
- Version: main branch at 30/10/2023

### Description

Hello Team. 

A python error arises when trying to validate the trace data of the Autoware's logging_simulator example execution. However, *babeltrace* parses it without issues. The error is a DispatchError from the multimethod package. 

I don't know if it could be related, but when building Autoware with CARET, the *lidar_apollo_instance_segmentation* package was not build using caret-rclcpp. I checked it with the following command.
```bash
ros2 caret check_caret_rclcpp -w ./
```
Moreover, I had another issue because of bokeh. But I could solve it downgrading bokeh from version 3.3.0 to 2.4.1. 

### Steps to reproduce

1. Follow the [Source Installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/) guide to build Autoware.
2. Follow the [Manual Installation](https://tier4.github.io/caret_doc/main/installation/manual_installation/) guide to build CARET.
3. Follow [this guide](https://github.com/tier4/CARET_report/tree/main/sample_autoware) to analyze Autoware with CARET.
4. The error arises at [this step](https://github.com/tier4/CARET_report/tree/main/sample_autoware#validate-trace-data-optional), when validating trace data. The error also occurs if I try to create the analysis report following the guide.

### Commands

```shell
ros2 caret check_ctf -d ~/.ros/tracing/autoware_launch_trace_20231027-143038/
```

### Logs

```shell
Succeed to find record_cpp_impl. the C++ version will be used.
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.7', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/me/Projects/ros2_caret_ws/build/ros2caret/ros2caret/command/caret.py", line 32, in main
    return extension.main(args=args)
  File "/home/me/Projects/ros2_caret_ws/build/ros2caret/ros2caret/verb/check_ctf.py", line 36, in main
    Lttng(args.trace_dir)
  File "/home/me/Projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 284, in __init__
    data, events, begin, end = self._parse_lttng_data(
  File "/home/me/Projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/lttng.py", line 307, in _parse_lttng_data
    data = Ros2DataModel()
  File "/home/me/Projects/ros2_caret_ws/build/caret_analyze/caret_analyze/infra/lttng/ros2_tracing/data_model.py", line 99, in __init__
    self.callback_start_instances = RecordsFactory.create_instance(
  File "/usr/local/lib/python3.10/dist-packages/multimethod/__init__.py", line 326, in __call__
    func = self[tuple(func(arg) for func, arg in zip(self.type_checkers, args))]
  File "/usr/local/lib/python3.10/dist-packages/multimethod/__init__.py", line 320, in __missing__
    raise DispatchError(msg, types, keys)
multimethod.DispatchError: ('create_instance: 2 methods found', (<class 'NoneType'>, <class 'multimethod.<class 'list'>'>), [(<class 'multimethod.typing.Optional[collections.abc.Sequence[dict[str, int]]]'>, <class 'multimethod.typing.Optional[collections.abc.Sequence[caret_analyze.record.column.ColumnValue]]'>), (<class 'multimethod.typing.Optional[collections.abc.Sequence[caret_analyze.record.interface.RecordInterface]]'>, <class 'multimethod.typing.Optional[collections.abc.Sequence[caret_analyze.record.column.ColumnValue]]'>)])
```

xygyo77, 2023-10-30:
Thank you for your report. 
As you assumed, this error is caused by multimethod not being version 1.10 or higher.
Please manually upgrade your version or bring your caret up to date.

If you bring caret up to date,
`ros2 caret check_caret_rclcpp -w . /`
is,
`ros2 caret check_caret_rclcpp . /`
" -w" is unnecessary.

`ros2 caret check_ctf -d ~/.ros/tracing/autoware_launch_trace_20231027-143038/ `
is,
`ros2 caret check_ctf ~/.ros/tracing/autoware_launch_trace_20231027-143038/ `
will become " -d" is not necessary.

At the same time, bokeh will be updated to 3.x or later.

We are not familiar with the issue of lidar_apollo_instance_segmentation not being built with caret-rclcpp.
Have you already applied the following 3 FAQs?
https://tier4.github.io/caret_doc/main/faq/known_issues/#build

m-alcon, 2023-10-31:
Thank you very much for your help.

I had the repository up to date but I did not think about the repos in the src/ folder, sorry for that. I updated them and the multimethod problem is solved. 

Before, I had only applied the first of the FAQs ([Conflicts of dependency on libtracetools.so](https://tier4.github.io/caret_doc/main/faq/known_issues/#conflicts-of-dependency-on-libtracetoolsso)), since only with this modification all the packages were built and I thought all was OK. Now I have also applied the last one ([SYSTEM's rclcpp is referred](https://tier4.github.io/caret_doc/main/faq/known_issues/#systems-rclcpp-is-referred)) and all packages are built **using caret-rclcpp** (checked with `ros2 caret check_caret_rclcpp ./`). However, I also tried to apply the second one ([Build using ament_cmake](https://tier4.github.io/caret_doc/main/faq/known_issues/#build-using-ament_cmake)) and the building did not finish. I can try to reproduce it and send the build error in case you want it, but all Autoware packages are built with caret-rclcpp and I had no errors with `ros2 caret check_ctf`, so it seems all is OK so far. 

Thank you again.

xygyo77, 2023-10-31:
I am glad it worked out.
Regarding the build, basically, I think it is a good idea to implement everything in the FAQ.
Since caret is affected by ros2_humble, any changes to ros2 may require additional action. We recommend that you also check this FAQ.

closed 2023-11-01 by xygyo77