tier4/caret #185: Cannot use the filter to leave out specific nodes or topics

REN-Tianchi, 2023-11-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Help wanted

### Step where the issue is related to

Recording

### Environments

    OS: Ubuntu 22.04.3 LTS
    ROS 2: Humble
    Memory: 64.0 GiB

### CARET version

    CARET: v0.3.3  main branch

### Target application

target application: autoware release/2023.10 branch

### Description

Although in the tutorial section I noticed that there are nodes/topics filter provided to leave out or just focus on some specific nodes and topics, as I use the CLI tool as described in the tutorial, the final tracing result still has nodes and topics that I intend to leave out, resulting in very high latency in data analysis using python APIs. I tried to use the echo to see if the nodes are correctly sent to the $CARET_IGNORE_NODES and they did.

### Steps to reproduce

Before start recording I wrote the command lines as is shown in the tutorial session, however, after recording, I found the nodes to leave out are still in the converted yaml file.

### Commands

```shell
ulimit -n 65536
source ~/ros2_caret_ws/install/local_setup.bash
export CARET_IGNORE_NODES="/rviz*: /autoware_api*: /awapi*: /transform_listener*: /system*: /default_ad_api*: /ro*: /aggregator_node*: /planning*: /control*: /localization*: /map*" 
export CARET_IGNORE_TOPICS="/external*: /api*: /awapi*: /diagnostics*: /system/*: /vehicle*: /carla*: /planning/scenario_planning/motion_velocity_smoother*:  /parameter_events: /clock: /planning*: /control*: /localization*"
export ROS_TRACE_DIR=~/autoware/evaluate
ros2 caret record -s record_file_name
```

h-suzuki-isp, 2023-11-27:
CARET provides filtering functions by setting certain environment variables (e.g. CARET_IGNORE_NODES) when the application is launched.
Setting environment variables must be done on the terminal where the application is launched.
If the issues is not resolve, please ask again. Additionally, you may find the following tutorial helpful:(https://tier4.github.io/caret_doc/main/tutorials/recording/)

ymski, 2024-09-24:
@REN-Tianchi 
Since we have not received a response for a long time and the status of the issue is unknown, we are closing it.

closed 2024-09-24 by ymski