tier4/caret #192: Tracing data always filtered to 0 events

ZiweiSong96, 2024-01-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Recording, Visualization

### Description

I follow the instruction about caret and caret report to trace the performance of Autoware.Universe with AWSIM. I can record the data and the size of the data files are correct. But when I started creating architecture file, the loading data is filtered to 0 events. And the jupyter-lab will outputs that all the records are lost.
[image]
[image]
[image]
[image]

ZiweiSong96, 2024-01-12:
I reinstall ros2 humble and caret again, and install the caret demo for testing, the tracing data of caret demo is also filtered to 0.
[image]

ymski, 2024-01-15:
Hi @ZiweiSong96.

It appears that the trace data is lost.
Please refer to the following link.

- lttng document: https://lttng.org/docs/v2.13/#doc-channel-overwrite-mode-vs-discard-mode
- caret document: https://tier4.github.io/caret_doc/latest/recording/validating/#tracer-discarded

ZiweiSong96, 2024-01-15:
> Hi @ZiweiSong96.
> 
> It appears that the trace data is lost. Please refer to the following link.
> 
>     * lttng document: https://lttng.org/docs/v2.13/#doc-channel-overwrite-mode-vs-discard-mode
> 
>     * caret document: https://tier4.github.io/caret_doc/latest/recording/validating/#tracer-discarded
Thank you for your kind help, I followed the mentioned documents and increased the size of ring-buffer in CARET (8*4096), but the issues are still appeared. When I visualized the latency of node chain with message flow for caret demo, warnings appeared and the figure seems not correct.
[image]
[image]
[image]
Actually when I started the Autoware AWSIM tracing and caret demo tracing, I applied the trace filter at the beginning. But it doesn't seem to work (I use the CLI command to create the architecture file and the warnings appeared.).
[image]
[image]
[image]

ThankPan, 2024-01-29:
Hi. I'm also working on applying CARET on Autoware and encountered a lot of bugs. I didn't realize some of the problems may caused by the 'filtered to 0 events'. So thank you for the post. Have you solved the problem of filtering events and gained some meaningful results? They seem to be not very active and responsive to the issues. I figure we could work on this together.

ymski, 2024-01-29:
Hi @ZiweiSong96 
I apologize for the delayed response.  I will address each issue separately.

- `filtered to 0 events`
  - This is a bug in v0.4.23.
  - The problem can be avoided by downgrading to v0.4.22.

- Message flow not rendering correctly
  - It seems that the data link between the first and second nodes is not functioning properly.
  - The available DDS implementations are FastDDS or CycloneDDS.  If you are using a different DDS, analysis might not be possible. If you could share the trace data from the tutorial, we can confirm if the necessary events are recorded.

- Warning: `Cannot generate a safe runtime search path`
  - The issue seems to occur when referencing the system's rclcpp instead of Caret's rclcpp.
  - If the build is done following the links below, warnings can be ignored.
    - [Build Reference](https://tier4.github.io/caret_doc/latest/faq/known_issues/#build)
    - [Systems rclcpp Reference](https://tier4.github.io/caret_doc/latest/faq/known_issues/#systems-rclcpp-is-referred)
    
If you have any further questions or need clarification, feel free to ask.

ZiweiSong96, 2024-01-29:
> Hi. I'm also working on applying CARET on Autoware and encountered a lot of bugs. I didn't realize some of the problems may caused by the 'filtered to 0 events'. So thank you for the post. Have you [...]

Actually during the last few weeks I tried to solve the problem, I copied the whole src files from another PC (where CARET is working well) and then CARET can work at sometimes, it seems the main branch of CARET cannot work well. But anoter person (He also works on CARET) build CARET main branch successfully and can trace the Autoware. I must say it is really confuse me
o(╥﹏╥)o

ymski, 2024-01-30:
@ZiweiSong96 
I would like to provide additional details regarding the issue of the "filtered to 0 events" message that is displayed during the creation of the Lttng object. It appears that although the output suggests that all data is lost, the internal data is still retained. The message is misleading and does not seem to affect functionality.

I appreciate your reporting. We have plans to address this bug in the near future.

ZiweiSong96, 2024-02-06:
> Hi @ZiweiSong96 I apologize for the delayed response. I will address each issue separately.
> 
> * `filtered to 0 events`
>   
>   * This is a bug in v0.4.23.
>   * The problem can be avoided by downgrading to ~v0.4.22~ v0.4.21.
> * Message flow not rendering correctly
>   
>   * It seems that the data link between the first and second nodes is not functioning properly.
>   * The available DDS implementations are FastDDS or CycloneDDS.  If you are using a different DDS, analysis might not be possible. If you could share the trace data from the tutorial, we can confirm if the necessary events are recorded.
> * Warning: `Cannot generate a safe runtime search path`
>   
>   * The issue seems to occur when referencing the system's rclcpp instead of Caret's rclcpp.
>   * If the build is done following the links below, warnings can be ignored.
>     
>     * [Build Reference](https://tier4.github.io/caret_doc/latest/faq/known_issues/#build)
>     * [Systems rclcpp Reference](https://tier4.github.io/caret_doc/latest/faq/known_issues/#systems-rclcpp-is-referred)
> 
> If you have any further questions or need clarification, feel free to ask.

Thank you very much for your help, I will try these methods to fix the caret on my device.

closed 2024-02-06 by ZiweiSong96