tier4/caret #201: chain latency about the first and last node can not be display

HaochunLiang, 2024-08-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Issue type

Bug

### Step where the issue is related to

Visualization

### Environments

OS: Ubuntu 22.04.1 LTS
Kernel version: 6.5.0-44-generic
ROS 2: Humble
CPU: Intel(R) i5-14600k
Memory: 48.0 GiB

### CARET version

CARET: v0.5.9
CARET_analyze: 3655d76afcf8e58d58c060bccc78f8b0949e0372
CARET_trace: 5d777a34d5d30e18a66a345566f9441bfa24cd0f

### Description

When I want to print out the Chain latency, the latency information of the first and last nodes cannot be displayed.
[image]

### Steps to reproduce

from caret_analyze.plot import chain_latency
from caret_analyze import Application, Architecture, Lttng
from bokeh.plotting import output_notebook, figure, show
output_notebook()

arch = Architecture('yaml', '/path/to/architecture_file')
lttng = Lttng('/path/to/trace_data')
app = Application(arch, lttng)
path = app.get_path('target_path')

chain_latency(path, granularity='node', lstrip_s=1, rstrip_s=1)

ymski, 2024-08-16:
Hi @HaochunLiang . Thanks for asking the question.
By default, the latency of the nodes at the end of a path is not output, since the definition of paths in CARET does not allow for the natural definition of latency at the first and last nodes of a path. However, an option allows latency to be displayed for the first and last nodes as well.

The following is sample code for outputting latency at the end of a path.
```
from caret_analyze.plot import chain_latency
path = app.get_path('target_path')
path.include_first_callback = True
path.include_last_callback = True
chain_latency(path, granularity='node', lstrip_s=1, rstrip_s=1)
```

Result
[image]

Documentation on these options can be found [here](https://tier4.github.io/caret_doc/latest/configuration/inter_node_data_path/#considering-firstlast-callback).

Related links
- [Path Definition](https://tier4.github.io/caret_doc/latest/design/event_and_latency_definitions/path/)
- [Node Definition](https://tier4.github.io/caret_doc/latest/design/event_and_latency_definitions/node/)
- [Communication Definition](https://tier4.github.io/caret_doc/latest/design/event_and_latency_definitions/communication/)

ymski, 2024-09-24:
@HaochunLiang 
Since we have not received a response for a long time and the status of the issue is unknown, we are closing it.
If you encounter similar problems, please create another issue.

closed 2024-09-24 by ymski