tier4/caret #205: Does CARET work for Jazzy?

RaduLucianR, 2024-10-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/tier4/caret/blob/main/.github/CONTRIBUTING.md).
- [X] I've read [document](https://tier4.github.io/CARET_doc/main/) and no solution were found.

### Step where the question is related to

Setup

### Description

Hi, I've previously used CARET for ROS2 Humble, but now I'd like to use it for Jazzy. Has anyone tested if it works fine for Jazzy? The documentation mentions only Humble and Iron as the supported versions.

ymski, 2024-10-15:
@RaduLucianR 
Thank you for your question. We are preparing CARET to work in the jazzy. We are aware that some trace points used in CARET do not work in jazzy. Customizing CARET for jazzy will require changes to several repositories and will take some time to develop. We are targeting a release in 3 months. We kindly ask for your continued patience.

RaduLucianR, 2024-10-21:
Thank you for your answer @ymski! I understand the development takes time. I am just curious if CARET would somewhat work on Jazzy out-of-the-box, not necessarily having everything working perfectly.

closed 2025-03-03 by ymski